xlyric/PV-discharge-Dimmer-AC-Dimmer-KIT-Robotdyn #3: Pin pour cooler en Standalone

dprslt, 2023-01-24:
Hello, 

Je me construit un dimmer en me basant grandement sur ton code (merci pour ca, pour le moment ça marche au top).

Le dimmer chauffe pas mal (version 24A en standalone du coup) et il se semblerai plus sage d'ajouter de la ventilation dans le boitier.

Mon problème c'est que je ne vois pas sur quel PIN du weemos brancher la commande du mofset pour le ventilateur, il me semble qu'il y'a un conflit pour la standalone. 

Dans `config.h` on a deux bloc pour le standalone

```c
//// configuration for Standalone boards ( personnalisation )
#ifdef  STANDALONE
#define outputPin  D5 
#define zerocross  D6 // for boards with CHANGEBLE input pins
#endif
```
et plus bas

```c
//// Dallas Configuration for Standalone boards ( personnalisation )
#ifdef  STANDALONE
#define ONE_WIRE_BUS D7
#define RELAY1 D3  // Spécial pour relay commandé 
#define RELAY2 D2  //
#define COOLER  D5   /// 0 : off  -> 1 : On --> need a dry contact or opto
//// pins for cooler
```

Mais du coup le D5 est déjà utilisé pour l'output du Dimmer.
On est d'accord qu'il faut configurer un autre pin ?

Question auxiliaire du coup, pour quel usage avait tu prévu les deux relais ? Je n'ai rien vu dans les différentes docs

xlyric, 2023-01-24:
Salut, 

effectivement, je l'avais oublié celui la.

tu peux effectivement utiliser 1 des 2 relais dispo ( j'ai pas encore fait la doc sur le site ) 
/relay1=x 

0 - off
1 - on 
2 - switch ( on vers off ou inversement ) 

Je note ta remarque et je corrige ça à la prochaine mise à jour.

au passage pour en discuter : https://forum-photovoltaique.fr/viewtopic.php?f=110&t=41777

xlyric, 2023-01-24:
c'est corrigé
#define COOLER  D8

closed 2023-01-24 by xlyric