xlyric/PV-discharge-Dimmer-AC-Dimmer-KIT-Robotdyn #13: Not work littlefs in ESP32

first-leon, 2023-07-22:
Hello. I flash ESP32 firmware. Download and flash https://ota.apper-solaire.org/firmware/littlefs-dimmer.bin.
[image]

Reboot device, after i go http://IP and see:
[image]

I flash last version fimware (commit c52445833cbea1cb377228e219172822a69911e0)

xlyric, 2023-07-22:
Hi, 

with the ESP32, you must use this firmware for filesystem : https://ota.apper-solaire.org/firmware/spiffs-dimmer.bin 

it's not the same filesystem between ESP32 and ESP8266 . 

other things, I have lot of issues with ESP32, and everybody use ESP8266 ( it's work perfect and cheaper )

first-leon, 2023-07-23:
Hello Cyril.
It doesn't work for me anyway.
For the test, i use ESP32_Devkitc_v4 (https://aliexpress.ru/item/1005005727909266.html) board and use the repository version c52445833cbea1cb377228e219172822a69911e0 (last commit in master).
I tried to upload a file https://ota.apper-solaire.org/firmware/spiffs-dimmer.bin but it doesn't work. I also tried using the build target "Upload Filesystem Image" over USB and it doesn't work either :(

"ESP8266 ( it's work perfect and cheaper )" - I agree with you. But I want to have a board that works not over wi-fi, but over ethernet. This board is also great: WT32-ETH01 (although more expensive).

xlyric, 2023-07-23:
Ok, I know why it's not working, 
it's use the same 
        { "path": "firmware/partitions.bin", "offset": 32768 },
        { "path": "firmware/boot_app0.bin", "offset": 57344 },

between dimmer and router and I made a partition modification on router. 
I'm on holidays, you need to use VS code and compile all. 
I will make the modification when I'm back

xlyric, 2023-07-23:
[spiffs.zip](https://github.com/xlyric/PV-discharge-Dimmer-AC-Dimmer-KIT-Robotdyn/files/12139009/spiffs.zip)
I think I have found, try this 
1 line is missing in plateformIO.ini 
board_build.filesystem = spiffs

xlyric, 2023-07-23:
I can try, it's look good, 
-> https://ota.apper-solaire.org/ota.php 
-> ESP32 + Dimmer Robotdyn

first-leon, 2023-07-23:
Great! I will check it tomorrow or the day after tomorrow on the ESP32_Devkitc_v4 board. Then I will check on the WT32-ETH01 board.

xlyric, 2023-07-23:
ok, I will wait your feed back :D

first-leon, 2023-07-24:
I checked ESP32_Devkitc_v . It works!
I will check WT32-ETH01.

first-leon, 2023-08-03:
Everything is working now. I'm closing the question.

closed 2023-08-03 by first-leon

xlyric, 2023-08-03:
> <xlyric/PV-discharge-Dimmer-AC-Dimmer-KIT-Robotdyn/issues/13/1663202633@
> github.com>

Ok cool, i will check your commit when i m back