xlyric/PV-discharge-Dimmer-AC-Dimmer-KIT-Robotdyn #22: redémarrage ventilateur

Dom1066, 2023-09-29:
Aprés la chauffe, quand la température repasse en dessous de la température de consigne (en mode normal) et sans routage le ventilateur se met en  en marche jusqu'à la fin de la journée.

xlyric, 2023-09-29:
ok, je vais regarder

xlyric, 2023-09-29:
je pense avoir trouvé, 
tu as une consigne minimale configurée à 0 ? ( ou elle est à combien ? )

Dom1066, 2023-09-29:
la consigne de puissance  mini ? elle est  à 5. 

( je viens de contrôler , ça le fait pas en mode minuteur d'appoint)

xlyric, 2023-09-29:
ok, ba une coquille doit trainer alors qq part concernant l'ancien code de gestion du ventillateur

xlyric, 2023-09-29:
je viens de faire des tests complémentaire: 
le ventilateur se remet en route que dans le cas où : la consigne a été dépassée et si dans la plage de refroidissement ( donc entre la valeur max temp et les -10% ) il y a de nouveau une commande de routage.

Dom1066, 2023-09-29:
pourtant , je ne vois pas de routage pendant cette phase.

xlyric, 2023-09-29:
le routage se fait pas, mais une commande est quand même passée. et non prise en compte sauf pour le ventillo

Dom1066, 2023-09-30:
Bon , 
je pense avoir une piste.

D'après Home assistant le ventilateur est déclenché/arrêté quand l'entité "usable-xxx" passe au dessus et en dessous de 0.
Il faudrait qu'il soit déclenché par l'entité "Watt".

cela pose aussi des problèmes de démarrage de ventilateur la nuit en mode minuteur,pour moi la source du problème est la même.

Bon week-end.

xlyric, 2023-09-30:
ok, comme tu contrôles ton dimmer par MQTT ? 
je faisais des tests avec les requetes HTTP

Dom1066, 2023-10-01:
Non,
HA était juste pour moi un moyen d'essayer de comprendre ce qui ne va pas, le dimmer est contrôlé par le routeur.
Alors, fausse piste.
.

Dom1066, 2023-10-07:
j'ai réussi a faire fonctionner le ventilateur de façon logique en virant la task cooler et en écrivant les démarrages/arrêts et remontées HA dan main.cpp.
Par contre, j'ai pas su mettre un tempo de 15 secondes pour les arrêts.
Cela fonctionne en "normal" et "minuteur" avec ma config 1 deul dimer distant.pour les autres, je sais pas.
j'ai réessayer de passer par une task cooler mais pas réussi...

xlyric, 2023-10-07:
oui mais non. 
on ne touche pas à la task, 
le fait de faire tourner des services est pour éviter certains oublis et autre. 
je suis entrain de voir de toute façon le problème, j'ai passé la task à 30s et je j'étudie le fait qu'un ventilateur tourne au moins 1min voir plus encore après une charge.

xlyric, 2023-10-07:
c'est bon j'ai corrigé le bug et ajouté une amélioration.
c était en fait une bricole... juste à rajouter dans les conditions de démarrage que (&& security == 0)

j'en ai profité pour rajouter un timer sur l'extinction ( cooldown ) de 1 min

closed 2023-10-07 by xlyric

Dom1066, 2023-10-08:
Merci.

Dom1066, 2023-10-09:
Ce matin, le ventilateur en mode minuteur d'appoint n'a pas fonctionné.