xlyric/PV-discharge-Dimmer-AC-Dimmer-KIT-Robotdyn #28: le ventilateur ne fonctionne pas en mode minuteur d'appoint.

Dom1066, 2023-10-09:
Cette nuit le ventilateur n'a pas fonctionné en mode minuteur.
après un nouvel essais à l'instant (10h30), il a fonctionné 1 minute 30 puis s'est arrêté.

xlyric, 2023-10-09:
interessant, j'ai ma petite idée, je vais corriger

xlyric, 2023-10-10:
tests : sans miniteur, mise d'un consigne, le ventillo reste en fonctionnement
puissance à 0 : mise en route du minuteur : RAS, le ventillo reste en fonctionnement.
relance d'une commande à 0, pendant la minuterie, le ventillateur fini par s'arreter. -> NOK
relance d'une commande à >0, pendant la minuterie, le ventillateur reste en fonctionnement -> RAS

du coup j'ai identifié le problème, je vais corriger ce soir

xlyric, 2023-10-10:
corrigé

closed 2023-10-10 by xlyric