xlyric/PV-discharge-Dimmer-AC-Dimmer-KIT-Robotdyn #41: Limite de température non-respectée

brenard, 2023-11-22:
Un de mes dimmers (firmware 20231120 SSR-ZC) n'a pas respecté aujourd'hui sa limite de 70°C, montant la température de mon chauffe-eau jusqu'à plus de 90°C. Ça a déclenché la thermistance du chauffe-eau (heureusement) et coupé son alimentation ainsi que celles des dimmers.

Dans HA, je vois que le dimmer as continué à suivre les ordres de puissance envoyés par le routeur (et routés par le 1er dimmer en amont) jusqu'à la coupure :
[image]
[image]

En parallèle, je vois que la remontée de température dans HA était dans les choux depuis un bout de temps (ça a certainement un lien) même si à priori le dimmer devait lui bien la connaître, car sa minuterie d'appoint s'est bien déclenchée ce matin (configurée entre 6h15 et 7h15 avec une température de consigne de 45°C) :

[image]

xlyric, 2023-11-22:
je viens de regarder et il y a bien un bug, le dimmer s 'éteins à la consigne t° mais se remet en route dès qu'une consigne de puissance arrive de nouveau

xlyric, 2023-11-22:
c'est corrigé, j'ai rajouté 2 sécurités, 1 sur la commande et une sur la non réponse de la sonde

closed 2023-11-22 by xlyric

brenard, 2023-11-22:
Effectivement, j'ai l'impression que la sonde y est pour quelque chose : j'ai branché temporairement le dimmer sur une batterie usb pour suivre la retombé de température via la sonde du dimmer et après avoir fonctionné pendant quelques heures, plus rien ne remonte. Sur l'interface Web, la sonde est à 20°C (qui semble être une valeur par défaut) et dans les logs, j'ai un message `Dallas not connected`. J'ai attendu pour voir si ça remontait, mais rien et un reboot logiciel ne change rien. Après un reboot matériel, la sonde revient et remonte une valeur cohérente, mais au bout de quelques minutes, la sonde s'est déconnectée à nouveau.

Pour revenir sur les sécurités, je pense que si la sonde est HS, il ne faudrait pas chauffer et ne pas remonté de fausse valeur à 20°C. Idéalement il faudrait aussi remonter une température à zéro via MQTT pour qu'on puisse facilement voir dans HA qu'il y a un problème (voir même se configurer une automatisation et une notification).

xlyric, 2023-11-22:
> <xlyric/PV-discharge-Dimmer-AC-Dimmer-KIT-Robotdyn/issues/41/1823463020@
> github.com>

oui, je vais changer ça pour que ça soit plus parlant.

brenard, 2023-11-22:
Après avoir compris qu'il fallait que je renclenche manuellement ma thermistance pour quelle ne bloc plus l'alimentation du chauffe eau et des dimmers, j'ai eut un comportement bizarre. Un des dimmers ne s'allumait pas mais le SSR dessus était lui ouvert à 100%. En branchant un câble usb dessus, le dimmer s'allumait correctement et le SSR s'etaignait, mais dès que je le debranchais, le dimmer n'était plus joignable et le SSR se rallumait. J'ai par ailleurs testé de débranché le wemos de la carte du Dimmer et le SSR s'etaignait. J'ai finis par tenté d'inverser les wemos de mes deux dimmers et les deux fonctionnaient correctement... Après les avoir remis à leur place, ils fonctionnent également tous les deux correctement. Étrange mais c'est un peu flippant de voir le SSR tourné "sans contrôle". Tu n'utiliserais pas par hasard un GPIO qui serait à UP par défaut au boot sur le wemos?

Tout est reparti pour le moment et j'ai mis à jour mes deux dimmers en espérant que ça se stabilise.

xlyric, 2023-11-23:
> <xlyric/PV-discharge-Dimmer-AC-Dimmer-KIT-Robotdyn/issues/41/1823666622@

étrange,
en principe toute les pins sont forcé à low a démarrage. c'est assez
étrange que le fait de retirer le wemos et le remettre résolve le problème.
au passage pour la dallas j'en ai jamais vu de HS, c'est assez étonnant.
donc soit la carte déconne, ce qui pourrait arriver, soit la sonde est mal
branchée

Si besoin je peux te fournir une nouvelle carte.

brenard, 2023-11-23:
Hum, de mémoire, il y a un certain nombre de GPIO qui sont UP au boot et après vérification, c'est effectivement le cas :
- GPIO16
- GPIO3
- GPIO1
- GPIO10
- GPIO9

Par ailleurs, les autres pins, sauf 4 et 5, peuvent avoir un peu de jus au boot également. Donc au final, faudrait idéalement brancher le SSR (et j'imagine le robotdyn) sur les pin 4 ou 5 pour éviter tout problème. 

https://randomnerdtutorials.com/esp8266-pinout-reference-gpios/

xlyric, 2023-11-23:
> <xlyric/PV-discharge-Dimmer-AC-Dimmer-KIT-Robotdyn/issues/41/1823959834@

oui, me souvient de ça.

Le SSR est en D1 donc en principe pas de problème

D1 GPIO5 OK OK often used as SCL (I2C)par contre ça mange pas de pain de
forcer de toute façon dès le début un état bas sur la pin.

je vais rajouter ça

xlyric, 2023-11-23:
c'est mis à jour

brenard, 2023-11-23:
> c'est mis à jour

Top, c'est déployé !