xlyric/PV-discharge-Dimmer-AC-Dimmer-KIT-Robotdyn #42: Valeur Watt et Watt total incorrect sur HA ?

multinet33, 2023-11-24:
Je me demande a quoi correspond les valeur Watt et Watt total sur Home Assistant car pour un dimmer de 1000 w de charge activé a 50% j'ai Watt a 0 et Watt Total a 50

[image]
[image]

xlyric, 2023-11-24:
étrange,
je regarderais demain.
tu utilises quel version compilé ? alone,SSR , etc?

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Sans
virus.www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

multinet33, 2023-11-24:
Je compile moi même et j’utilise un robotydn avec esp

Sans doute qu’il n’y a pas de problème et que je ne maîtrise pas à quoi correspond watt et watt total

closed 2024-04-12 by xlyric