xlyric/PV-discharge-Dimmer-AC-Dimmer-KIT-Robotdyn #43: Dallas not connected

multinet33, 2023-11-25:
Suite a mise a jour firmware de juillet vers le denrier de novembre
 la sonde dallas  branché sur les pin 12 14 et 02 du dimmer 2022 robotdyn n'est pas reconnu

un rollback sur le code de juillet permet d'avoir la sonde reconnue

xlyric, 2023-11-25:
il me semble que j'avais remis correctement les noms, t'as testé l'autre version robotdyn ?

multinet33, 2023-11-25:
malheureusement non je ne peux pas tester l'autre (avec ESP onboard et pas en carte fille) est mon dimmer de prod, je veux pas prendre de risque

multinet33, 2023-11-26:
Pour info, si je rajoute ce bloque juste avant le dallaspresent(); dans setup() du main.cpp ca marche.

Comme cela est deja fait plus haut il y a qqch qui le reinitialise avant le dallaspresent()
```
  #ifdef POWERSUPPLY2022  
  pinMode(GND_PIN, OUTPUT);  /// board bug
  digitalWrite(GND_PIN, 0);  /// board bug with pin 16 

  pinMode(POS_PIN, OUTPUT); 
  digitalWrite(POS_PIN, 1);
  #endif
```

xlyric, 2023-11-26:
ça y est je pense savoir pourquoi, ( entre temps ) 
vu que SSR et autre, j'ai fait un bloc de code pour unifier les commandes et que ça soit plus simple coté commandes

après rien n'empêche de rajouter ce bout de code.

multinet33, 2023-11-26:
tu le rajoutes sans que je fasse un pull request ?

je me pose plein de questions sur pourquoi le dimmer remet a 0 le power dans les qq secondes qui suivent un ordre ?/POWER=xx => sur le fourm, je suis preneur de tes avis

xlyric, 2023-11-26:
j'ai répondu sur le forum ;)

xlyric, 2023-11-26:
la modif est push