xlyric/PV-discharge-Dimmer-AC-Dimmer-KIT-Robotdyn #44: phase de descente de température de 10% mode minuteur.

Dom1066, 2023-11-30:
Bonjour,
en mode minuteur, si la température de consigne est atteinte avant la fin de la période de programme,
la "chauffe" redémarre dès la chute te température de 0,1 °C en dessous de la consigne (et pas chute de 10%).

C'est peut-être voulu pour avoir la température max en fin de programme ?

Vu la période et la météo, impossible de dire si c'est pareil en mode "standard".

xlyric, 2023-11-30:
en fait c'est ce qu'on m'a demandé dernièrement donc oui c'est normal.

xlyric, 2023-12-01:
je vais quand même regarder un peu pour éviter que ça bagotte
il y a aussi le fait que j'ai rajouté qq rêgle pour qu'en cas de reboot, le minuteur reprenne et ça peut jouer

Dom1066, 2024-03-06:
bonjour,
j'ai contourné en mettants des automatisations dans HA.
à 02h00 
si Temp entre 51 et 60 Pmax dimmer= 35
si Temp entre 40 et 50 Pmax dimmer= 57
à 07h10 remise Pmax par défaut= 75
avec programmation de 03h00 à 06h57.
ça prolonge le temp de chauffe à puissance moindre et limite les yoyo en fin de chauffe.
je peux affinner en mettant plus de plages de température mais c'est déjà pas mal.

xlyric, 2024-03-06:
> <xlyric/PV-discharge-Dimmer-AC-Dimmer-KIT-Robotdyn/issues/44/1980324322@
> github.com>

Je suis entrain de corriger certains petits bugs, mais j'aime bien l'idée.
je verrais si ça plait aux autres utilisateurs

Dom1066, 2024-04-09:
les valeurs des automatisations dans HA ne commandent plus le dimmer.

closed 2024-04-27 by Dom1066