xlyric/PV-discharge-Dimmer-AC-Dimmer-KIT-Robotdyn #46: heure page minuteur

Dom1066, 2024-01-13:
Bonjour et bonne année.
Depuis quelques temps (passage à 2024 ?) l'heure sur la page minuteur est décalée de -1heure / à l'heure réelle et l'heure de la page index.
la programmation est alors décalée de 1heure.
(testé evec dernière version du git et c'est pareil)

xlyric, 2024-01-13:
Salut, 

étrange, je regarderais. 
je suis sur la version du 23/11 et j'ai pas constaté le problème. ça apparait aussi sur l'heure de la page le décalage, ou c'est juste au déclenchement ?

Dom1066, 2024-01-14:
ça apparait aussi sur l'heure de la page minuteur, la page index est juste.laprogrammation est décalée.

Dom1066, 2024-01-14:
étrange !!! je viens de regarder par hasard (car je ne l'utilise pas) la page minuteur de mon PV routeur a aussi 1 heure de décalage !!!

xlyric, 2024-01-16:
je viens de corriger sur le dimmer, je vais reporter la librairie sur le routeur

closed 2024-01-16 by xlyric