xlyric/PV-discharge-Dimmer-AC-Dimmer-KIT-Robotdyn #51: Fonctionnement sans internet

Pmachat, 2024-04-11:
Bonjour
Est il possible de faire fonctionner le dimmer sans accès internet mais uniquement avec un réseau local?
Je vois des appels à certains sites internet comme 
Ota.apper-solaire.org / Ajax.googleapis.com/code.jquery.com
Quand je me connecte à l interface local web en 192.x.x.x

au final, cela donne des morceaux de l interface incohérent 

merci d avance

xlyric, 2024-04-11:
j'ai effectivement commencé à faire ce qu'il faut pour que ça se fasse , j'ai déjà bien avancé sur le routeur.
pour le dimmer, je ferais comme le routeur, à l'interieur de la page HTML, 
si la page n'est pas trouvée, qu'il ailles prendre sur le FS ( si il reste de la place . )  ( pour le Jquery ) 
après pour ce qui est du contenu vers OTA ou ko-fi, c'est pas important, 

je vais tenter de faire ça dans la semaine

xlyric, 2024-04-12:
Modif faite, il faut test et me dire ce qui va pas.

Pmachat, 2024-04-12:
Désolé, je ne suis pas très doué. Je ne sais pas créer l'executable ( fichier bin) à partir des fichiers sources modifiés
Je vais regarder mais cela va prendre du temps

xlyric, 2024-04-13:
ça se fait avec VS code : 
https://pvrouteur.apper-solaire.org/books/pv-routeur-et-dimmer-numerique/page/90-mode-dev-installation-de-visual-studio-code
[firmware.zip](https://github.com/xlyric/PV-discharge-Dimmer-AC-Dimmer-KIT-Robotdyn/files/14965618/firmware.zip)

xlyric, 2024-04-18:
Je n'ai pas eu de retour , est ce que c'est bon ?

closed 2024-04-24 by xlyric

Pmachat, 2024-04-29:
Bonjour,
Après mon retour de vacances, les conclusions sont que cela fonctionne pour la parie configuration en réseau local ( pas de visualisation et de mqtt)
Merci pour le service !