xlyric/PV-discharge-Dimmer-AC-Dimmer-KIT-Robotdyn #53: Microbug : MQTT décoché et les log parle de "connecting to MQTT"

ciscolyon69, 2024-05-23:
Salut,

J'ai détecté un microbug sur la version [Dimmer JOTTA ZC](https://ota.apper-solaire.org/firmware/firmwareJOTTA.bin)  20240501 SSR-ZC

Le mqtt est décoché et les log parle de  "connecting to MQTT"

"hostname":"","port":1883,"Publish":"domoticz/in","IDXTemp":200,"maxtemp":50,"IDXAlarme":202,"IDX":201,"startingpow":0,"minpow":0,"maxpow":90,"child":"192.168.1.90","mode":"equal","SubscribePV":"none","SubscribeTEMP":"none","dimmer_on_off":true,"charge":1200,"HA":true,"JEEDOM":true,"DOMOTICZ":true,"PVROUTER":"mqtt","DALLAS":"2861640af3493dfc","name":"Dimmer-88AE","charge1":0,"charge2":1200,"charge3":0

[image]

Bernex sur le forum

xlyric, 2024-05-23:
je vais tester et corriger, ça doit pas être grand chose

xlyric, 2024-05-23:
rectification, 
j'avais vu ça l'autre fois lors d'une install sur Bordeaux ( avec Vevere ) et ça a été corrigé depuis

closed 2024-05-23 by xlyric

ciscolyon69, 2024-05-23:
encore merci