xlyric/PV-discharge-Dimmer-AC-Dimmer-KIT-Robotdyn #76: température de consigne /reboot,

Dom1066, 2025-01-26:
Bonjour,
 en dernière version compilée du git 2 soucis apparaissent chez moi :
- La températue de consigne par défaut (50) est appliquée pour le minuteur même si une autre est enregistrée.
- Si on arrête le "Boost" manuellement avec un click sur l'interface web, le dimmer reboot.(tous les autres cas de figure foncionnent (déroulement complet;HA)

xlyric, 2025-01-26:
Salut Dom, 

c'est noté, je vais regarder ça, c'est étrange. 
après pour la température du minuteur, il prends la température par défaut si la température du minuteur est supérieure ( pour raison de sécurité ) , c'est le cas ?

xlyric, 2025-01-26:
c'est une bonne piste, je vais regarder

xlyric, 2025-01-26:
effectivement, ça reboot à la coupure.
c'est un conflit entre les envoies MQTT et HTTP.

pour la consigne par défaut, il faut que je test ça en condition réelle, il est possible que les 2 bugs soient liées. si tu peux test ?

Dom1066, 2025-01-26:
ok pour le reboot , ça fonctionne.

xlyric, 2025-01-26:
Je tenterais de voir demain pour tester pour cette histoire de limite à 50°
comme pas beaucoup de soleil en ce moment

Dom1066, 2025-01-27:
Je viens de faire un essai pour la tempérarure et cela ma parait OK. Cela devait être lié.

xlyric, 2025-01-27:
pareil sur mes graph, j'avais fait chauffé à  60 et ça semble être bon

closed 2025-01-27 by xlyric